JingOS-team/JingOS #1: Why not make the OS OpenSource and make the DE available on other distros?

TamrielNetwork, 2021-02-01:
This OS would definitely be a great alternative for mobile devices. At least layout-wise. But only downloading a whole distro because it looks good imo isn't really the intended way to choose an os. Else you could also just buy an ipad instead of using non-optimized linux software for mobile devices.
If the OS would be open source it would be a great alternative and would be much more secure (see other opensource distros) because many skilled people can improve it!
Also not even being able to use only the DE is even worse than deepins closed source software.
I don't really know what the intention with this is, but if it's not open source there's basically no reason to use it.. it's in no way better than IOS and you can't even tell how much tracking is going on here..
Would be very good for the company and the community to make this opensource, else I don't see a future for it.

JingOS-team, 2021-02-03:
We are making preparations before open source, and we expect to upload the first batch of source code on github next week. Thank you for your issue!

closed 2021-02-03 by JingOS-team

Drugantibus, 2021-03-31:
Almost two months have gone by and there's still no source? Isn't this a breach of the Linux license?

TamrielNetwork, 2021-03-31:
> Almost two months have gone by and there's still no source? Isn't this a breach of the Linux license?

I don't know about the license breach to be honest. I at least don't think it is.

Also in the README.md they wrote this:

```
About open sourcing:

> JingOS is based on Ubuntu 20.04, KDE v5.75, Plasma Mobile 5.20. We will replace the framework from Plasma Mobile to JDE (Jing Desktop Environment) later this year.
```

Probably meaning that they will migrate to their own desktop environment and open-source that.
One thing that'd be nice however is to have a version that you can build from source (or build and install on top of Debian/Ubuntu), either with the new desktop environment or plasma.

If they replace kde and plasma mobile with Jing DE then you should be able to just set up a Ubuntu (or Debian?) distro and install Jing DE on top of that.

TamrielNetwork, 2021-03-31:
Also since distro/de changes (cosmetic or others) are mostly in a human readable config it'd be nice to open source those too.
Package lists and other changes to the distro too ofc.

I think it's just easier to wait until they release their own DE with open sourcing the cosmetic changes.

And ofc, making the distro an official open source fork of Ubuntu and got the package(s) for the DE in the official repos (maybe also snap or whatever) would be easier if you want to have the developer intended experience (also regarding Tablets and ease of installation) but also be sure that it is open source and nothing unwanted is running in the background.