JingOS-team/JingOS #5: What do you think of JingPad A1?

JingOS-team, 2021-03-19:
It is the World’s FIRST Consumer-level Linux Tablet.

It is the World’s FIRST Linux Tablet with a 2K+ Display.

It is the World’s FIRST Linux Tablet with 5G/4G support.

It comes with the FIRST detachable Keyboard with a trackpad in Linux World.

It is the World’s FIRST Linux Tablet that comes with a Stylus.

It is the FIRST Tablet powered by JingOS – the World’s FIRST Linux Tablet OS.

It is the FIRST Time you can use a Linux-based Tablet as your daily driver.

It is the JingPad A1!

https://en.jingos.com/jingpad-a1/

Unnewehr, 2021-03-21:
Will it be available in Germany and how much will it cost?

nojd11, 2021-03-22:
It is not currently available for sale, and according to official sources, it may be available in June 2020, although it is not yet known whether it will be available in Germany.

pekinlcc, 2021-03-22:
will start a crowdfunding campaign by the end of May.

just-carlod, 2021-03-23:
Nice issue 😂

subjectdenied, 2021-03-26:
will it be possible with some effort from a user to run other desktop environments like the new gnome40?

pekinlcc, 2021-03-27:
> will it be possible with some effort from a user to run other desktop environments like the new gnome40?

I think yes. We'll involve the community to porting other OSs/DEs to the JingPad A1.

patrickjquinn, 2021-04-10:
I think the marketing for it needs a bit of work. This narrative that its the first Linux tablet with half of those features is patently untrue (as seen by a simple Google search)

To name a few:

- All of the ChromeOS tablets.
- HP TouchPad
- The Jolla Tablet
- The PineTab
- All of the RasPi tablets
- A ton of Linux tablets out of China
- The Librem 11
- The Linux Netvertibles from a decade ago.

The list goes on.

Also, avoid using GitHub issues like its twitter. Thats not what it's for and breaks the sanity of a repo. There is a dedicated discussion feature on GitHub for this sort of thing.

All of the above aside, i'm following the project along very closely and i'm very happy with the decision to open source the work you've done (which is very impressive btw.)

I'm rooting for you folks to bring an experience led option to mainline Linux devices. 🙂 Keep up the good work.

Kreyren, 2021-11-19:
**DISCLAIMER:** I do not own or received the device and i am submitting this using available info and as an act of good faith hoping that the developer takes action on the highlighted issues.

## Hardware

JingPad is a proprietary device as evident by the lack of publicly available ECAD files (that would enable anyone to import the tablet in ECAD software such as KiCAD and generate gerber files for home/pro production) and jingpad's inaction to release them (https://github.com/JingOS-team/JingOS/issues/67) which would enable the right to use, study, improve and redistribute (e.g. using CERN license).

As such it's impossible to verify whether the tablet ships with a supply-chain attack such as built-in backdoor to the kernel's ring-0 (which is common as evident by spectre and meltdown vulnerabilities) and it's design is impossible to fork and adapt to the user's needs. **RECOMMENDATION:** Release the ECAD files under mentioned license and/or submit the tablet for an independent security review.

For those reasons i discourage you from buying jingpad if you care about your:
1. privacy
2. security
3. The long-term functionality and longevity the device

With the lack of spare parts this device and lack effort to protect the fragile front display it's also very likely to contribute to the global warming crisis as it's evidently not made to last and i wasn't able to find any commitment by the developer to design the device to last. **RECOMMENDATION:** Commit to providing spare parts and files needed to make them e.g. OpenSCAD files in a repository to create the case.

### Display
#### Questionable impact protection
Looking on the pictures and videos of the device i am not convinced that edges of the display being at sufficient height to protect the glass from impact on a flat surface display first. **RECOMMENDATION:** Perform drop test from the height of >1.5m and adjust design as appropriate.

#### AMOLED Burn-ins
I am concerned about the pixel burn-ins (usage of digitizer, glue and/or glass that reacts to the light emitted from the display that causes a permanent "burn in" of the UI element to the display) that are common to AMOLED displays which should be clarified as with the UI of jingOS it's likely that this display will me sensitive to pixel burn-ins.

#### Lack of support for glass replacements
About replacements of glass: In most cases the glass cracking does not damages the digitizer or the display itself meaning that using heat, molybdenum wire and tape allows to cut the used (usually OCA) glue with the wire to release it from the digitizer assembly to replace the glass without the need to replace the whole display assembly which can be done by an inexperienced amateur (https://www.youtube.com/watch?v=xhSAmai8OKk). To make such repair easier:
1. the thickness of the glue layer should be adjusted with tolerance for a molybdenum wire
2. the glue used should be removable with a known chemical that is not reacting with the components around it
3. The glue installation should be designed to not require special tools
4. A form that holds the display assembly and allows easier alignment of the glass should be provided

### The USB-C connector
The connector has a limited set of uses before it's unable to hold the wire to maintain connection. To mitigate this issue the USB-C connector should be made modular to enable user replacement without the need of soldering.

### Case for hardware switches
Hardware switches to disable the camera, microphone and wifi/ethernet modem are essential to mitigate any software-related surveillance or malicious activity and should be implemented.

### Battery
The battery has a limited life-span after which the display has to be uninstalled and the battery removed. To avoid replacement that will most likely require a professional help the battery should be adapted to be hot-swapable. 

## Software
### Linux Distribution (JingOS)
I am not aware of the developer adapting the GNU's Free Software Distribution Guideliness (https://www.gnu.org/distros/free-system-distribution-guidelines.html) so it is likely that this device will promote the use of software that doesn't comply with four freedoms of free software which makes it impossible to verify whether those components are not performing malicious payload at the background such as (in one of the worst case scenarios) sending live video feed for telemetry to a remote server.

### SoC
The JingPad A1 currently uses UNISOC Tiger T7510 which doesn't seem to provide a libre bootloader and the microcode as such it is problematic to install any non-jingOS operating system without soldering of the storage chip (and even then it's likely that the tablet won't load the new OS) and it's impossible to verify what the microcode does which makes it impossible to rule out malicious activity.

I would suggest to consider using a manufacturer that committed to open-source such as rockchip (https://wikiless.org/wiki/Rockchip#Open-source_commitment) to mitigate these issues.

closed 2021-12-17 by JingOS-team