JingOS-team/JingOS #7: Possible instalation in a Xperia Tab

LucasDRavarotto, 2021-03-27:
Best regards, I would like to consult the possibility of installing the system on a Sony tablet, the Z2 tab. Is it possible that it runs with a Snapdragon 801 processor and being ARM 32 bits?

pekinlcc, 2021-03-29:
> Best regards, I would like to consult the possibility of installing the system on a Sony tablet, the Z2 tab. Is it possible that it runs with a Snapdragon 801 processor and being ARM 32 bits?

We only have JingOS for x86 by now. Maybe later, the community can help to port the ARM version to Snapdragon tablets.

AggamR, 2021-04-24:
> We only have JingOS for x86 by now. Maybe later, the community can help to port the ARM version to Snapdragon tablets.

Any project you know of that's trying to do so? Also, does that mean there is an ARM version?

closed 2021-12-17 by JingOS-team