JingOS-team/JingOS #35: Can't download

hunabku94, 2021-05-31:
https://hk.mikecrm.com/N0wgIm4 is unreachable.

JingOS-team, 2021-05-31:
It is still work, you can also try below links:
https://drive.google.com/file/d/1nstG1nAvxC4-U5qY_owAcptf_Y8LYTme/view?usp=sharing
https://download.jingos.com/os/JingOS-V0.8.1-ed3d91.iso

JingOS-team, 2021-05-31:
It still works, you can also try the below links:
https://drive.google.com/file/d/1nstG1nAvxC4-U5qY_owAcptf_Y8LYTme/view?usp=sharing
https://download.jingos.com/os/JingOS-V0.8.1-ed3d91.iso

GHNewbiee, 2021-06-01:
0.9 download link does not work.

Keep in mind that when you announce a new version and give a download link which does not work, then you disappoint people with all the consequences...

GHNewbiee, 2021-06-01:
1.Google driver:

https://drive.google.com/file/d/1l2-VMvozA_7DGulSIz8lWF7FMWYqRlYJ/view?usp=sharing

2. Download link: https://download.jingos.com/os/JingOS-V0.9-a25ea3.iso

hunabku94, 2021-06-08:
Thanks !

 
 

Envoyé: mardi 1 juin 2021 à 16:07

 

1.Google driver:

https://drive.google.com/file/d/1l2-VMvozA_7DGulSIz8lWF7FMWYqRlYJ/view?usp=sharing

	Download link: https://download.jingos.com/os/JingOS-V0.9-a25ea3.iso

closed 2021-06-21 by hunabku94