JingOS-team/JingOS #47: Qt Platform Plugin could not be initiated

cdqz2014, 2021-07-29:
I tried JingOS on my Thinkpad X1 Yoga gen3. I was truly astonished 'cause almost all function works as expected, except when I try to install some applications like `mathematica`, `netease-cloud-music`, and `xDroid` etc.

For example, when running `mathematica` (newest version 12.3), I get
```
Qt issued a fatal error: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, minimal, offscreen, vnc, xcb.

Qt tried to load plugins from:
/usr/local/Wolfram/Mathematica/12.3/SystemFiles/Libraries/Linux-x86-64/Qt/plugins
/usr/local/Wolfram/Mathematica/12.3/SystemFiles/FrontEnd/Binaries/Linux-x86-64

Mathematica was installed in /usr/local/Wolfram/Mathematica/12.3
Aborted (core dumped)
```

When running `netease-cloud-music` (ubuntu deb from official website), I get
```
This application failed to start because it could not find or load the Qt platform plugin "wayland"
in "/opt/netease/netease-cloud-music/plugins/platforms".

Available platform plugins are: eglfs (from /opt/netease/netease-cloud-music/plugins/platforms), linuxfb (from /opt/netease/netease-cloud-music/plugins/platforms), minimal (from /opt/netease/netease-cloud-music/plugins/platforms), minimalegl (from /opt/netease/netease-cloud-music/plugins/platforms), offscreen (from /opt/netease/netease-cloud-music/plugins/platforms), vnc (from /opt/netease/netease-cloud-music/plugins/platforms), xcb (from /opt/netease/netease-cloud-music/plugins/platforms), eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Reinstalling the application may fix this problem.
Received signal 6
#0 0x7f231eb65526 <unknown>
#1 0x7f231eb65909 <unknown>
#2 0x7f2324c28210 <unknown>
#3 0x7f2324c2818b gsignal
#4 0x7f2324c07859 abort
#5 0x7f2325085fa7 QMessageLogger::fatal()
#6 0x7f2325abf31a QGuiApplicationPrivate::createPlatformIntegration()
#7 0x7f2325abf41d QGuiApplicationPrivate::createEventDispatcher()
#8 0x7f2325280865 QCoreApplicationPrivate::init()
#9 0x7f2325ac0eaf QGuiApplicationPrivate::init()
#10 0x7f2327acf649 QApplicationPrivate::init()
#11 0x56166a62ad83 <unknown>
#12 0x56166a628d50 <unknown>
#13 0x7f2324c090b3 __libc_start_main
#14 0x56166a629efa <unknown>
  r8: 0000000000000000  r9: 00007ffc69023b00 r10: 0000000000000008 r11: 0000000000000246
 r12: 000056166c60c8d0 r13: 00007ffc69023f10 r14: 0000000000000018 r15: 00007ffc69023f08
  di: 0000000000000002  si: 00007ffc69023b00  bp: 00007ffc69023f18  bx: 00007f230f952d00
  dx: 0000000000000000  ax: 0000000000000000  cx: 00007f2324c2818b  sp: 00007ffc69023b00
  ip: 00007f2324c2818b efl: 0000000000000246 cgf: 002b000000000033 erf: 0000000000000000
 trp: 0000000000000000 msk: 0000000000000000 cr2: 0000000000000000
[end of stack trace]
Calling _exit(1). Core file will not be generated.
```

I googled for help, but almost all solutions asked to look at  the outputs of `ldd` (and then install the required pkg), which seems to be normal for me.

Is there any solution for this? Thx.

JingOS-team, 2021-07-29:
@cdqz2014 How did you install them? Did you run them in konsole?  You can check the "QT_QPA_PLATFORM" enviroment  in you shell, the default value is "wayland". If the app have it's own qt libs  and doesn't provide wayland qpa plugins, it will get a "not found" error. You can modify the enviorment by running “export QT_QPA_PLATFORM=xcb” in you shell ，and then try to run them again.

cdqz2014, 2021-07-29:
@JingOS-team Yes the default environment `QT_QPA_PLATFORM=wayland` is not working for `mathematica`. After export to `xcb` it runs! So does `netease-cloud-music`. Thanks.

BTW, since I set the global scaling of KDE to 100%, `netease-cloud-music` is not showing fullscreen. Are there any patch to it? 
[image]

closed 2021-07-30 by cdqz2014