JingOS-team/JingOS #65: Can't install nix package manager because /proc is busy

mogorman, 2021-11-03:
I am trying to install nix package manager on my jingpad .

when running the install it gets caught here
```
---- sudo execution ------------------------------------------------------------
I am executing:

    $ sudo HOME=/root /nix/store/h48lkygcqj4hdibbdnpl67q7ks6vkrd6-nix-2.4/bin/nix-env -i /nix/store/h48lkygcqj4hdibbdnpl67q7ks6vkrd6-nix-2.4

to install a bootstrapping Nix in to the default profile

installing 'nix-2.4'
error: mounting /proc: Device or resource busy
(use '--show-trace' to show detailed location information)

---- oh no! --------------------------------------------------------------------
```

**Steps To Reproduce**
Run `sh <(curl -L https://nixos.org/nix/install) --daemon` on a jingpad A1

I also reported here https://github.com/NixOS/nix/issues/5483

the software has no problem running on other ubuntu arm platforms or machines I have tried it on.

mogorman, 2021-11-04:
I tried adding the ubuntu impish package source as it has nix package manager as a package but if i installed it it would break jingos most likely

```
root@JingOS:~# apt-get install nix-bin 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  drihybris libexif12 libperl5.30 libqt5opengl5 libxfont2 perl-modules-5.30 repowerd-data sensorfw-qt5 x11-xkb-utils xserver-common
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  libboost-context1.74.0 libc-bin libc-dev-bin libc6 libc6-dev libdbd-sqlite3-perl libdbi-perl libgc1 liblocale-gettext-perl libnsl-dev libnsl2 libperl5.32
  libtext-charwidth-perl libtext-iconv-perl libtirpc-common libtirpc-dev libtirpc3 libwww-curl-perl locales perl perl-base perl-modules-5.32 rpcsvc-proto
Suggested packages:
  manpages glibc-doc manpages-dev libclone-perl libmldbm-perl libnet-daemon-perl libsql-statement-perl libcurl4-gnutls-dev perl-doc libterm-readline-gnu-perl
  | libterm-readline-perl-perl libtap-harness-archive-perl
Recommended packages:
  manpages manpages-dev libc-devtools libnss-nis libnss-nisplus nix-setup-systemd
The following packages will be REMOVED:
  cameraplugin-aal gstreamer1.0-droid gstreamer1.0-droid-tools kwin-wayland kwin-wayland-backend-hwcomposer libandroid-properties1 libhardware2 libhybris
  libhybris-common1 libhybris-test libhybris-utils libmedia1 plasma-workspace-wayland qtvideonode-plugin repowerd xwayland
The following NEW packages will be installed:
  libboost-context1.74.0 libdbd-sqlite3-perl libdbi-perl libgc1 libnsl-dev libnsl2 libperl5.32 libtirpc-dev libwww-curl-perl nix-bin perl-modules-5.32 rpcsvc-proto
The following packages will be upgraded:
  libc-bin libc-dev-bin libc6 libc6-dev liblocale-gettext-perl libtext-charwidth-perl libtext-iconv-perl libtirpc-common libtirpc3 locales perl perl-base
12 upgraded, 12 newly installed, 16 to remove and 782 not upgraded.
Need to get 21.9 MB of archives.
After this operation, 37.5 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

noaccOS, 2021-11-29:
Since the same bug appears on ChromeOS, and this distro uses the android kernel, can the problem be some modification to the kernel made by Google?