JingOS-team/JingOS #85: Battery draining issue while JingPad on standby mode.

mok888, 2021-12-12:
Battery draining while JingPad fully charged and put on standby mode.

Condition:
1. 100% charged.
2. Plane mode activated. (Wifi + BT off)

Kindly look into it as this will be a critical requirement for user especially on/ off using the pad.

JingOS-team, 2021-12-13:
We are optimising this issue at the moment. There might be some significant battery drain when connecting the keyboard or keeping the BT and WIFI on.

susanin-ivan, 2021-12-26:
Non-stop writting debug info into system log might be source of this issue. Just run "journalctl -f" from terminal - this flood never stops

JingOS-team, 2022-01-10:
I have assigned this issue to our test team

closed 2022-01-10 by JingOS-team

susanin-ivan, 2022-01-10:
Usually issue are closed when fix is available. 
Closing performace issue unfixed is a bad habbit and does not adds to your reputation.