JingOS-team/JingOS #95: Tooltips and open menus are visible once the tablet is locked

raquelhortab, 2021-12-16:
If you open a  dropdown menu or hover something that shows a tooltip and lock the tablet, the menu/tooltips is still visible once you turn the screen on again, on the lock screen. See this example:

[image]

JingOS-team, 2022-01-10:
这个可能是窗口管理问题？待确认

songlijuan666, 2022-01-10:
这个应该是三方应用的下拉框，自研的应用是没有的，终端下拉菜单也没有这个问题，可以问问用户安装的哪个应用

raquelhortab, 2022-01-10:
Hi, I believe it was okular or xournal, I'll try to check and confirm later today. It was an app from the jing app store almost for sure. Judging from the options in the menu, I'm pretty sure it must have been Xournal (see "XInput" and ".xoj" extension).

songlijuan666, 2022-01-11:
新版本该问题已经不复现，待1月发版版本请再看看

raquelhortab, 2022-01-11:
OK, I'll try! Thanks

raquelhortab, 2022-01-11:
I confirm that it was Xournal and that it still happens after running `sudo apt-get update`

songlijuan666, 2022-01-13:
等待新版本的时候，请再执行升级验证

songlijuan666, 2022-01-24:
新版本V1.2已经发布，请再次试试

raquelhortab, 2022-01-25:
Hi, I tried upgrading and my jingpad is now stuck on the logo screen... Can't try anything... can someone from jingpad help? I asked in telegram as well

songlijuan666, 2022-01-25:
强制重启

raquelhortab, 2022-01-25:
Of course I have tried this plenty of times... It's been like that for 9 or 10 hours already

songlijuan666, 2022-01-25:
这个情况目前暂时未遇到过，只能尝试复现；您那有重要数据需要保留的吗？如果重启一直起不来，可能就的恢复出厂值了

raquelhortab, 2022-01-25:
well it took me about two weeks to setup my whole work environment there so it would be quite a waste of time.... Of course, if there is no other way, I'll have to do it. Maybe I should wait for the battery to end so it turns off? Or that'd be worse and then I might not get it to turn on again?
Right now all it does is stay on the black screen with the logo, if I force it to turn off, it automatically turns on by its own, so I have no way of turning it off.

songlijuan666, 2022-01-25:
您再等等，再充充电等等

raquelhortab, 2022-01-25:
Hi, I managed to turn it off by unplugging it and pressing the power button for 15 seconds. I also managed to enter recovery mode and see these options in the picture. I'd like to preserve all my data and configuration, is there any way?

[image]

songlijuan666, 2022-01-26:
选择wipe data/factory reset，进入后，有个保留数据擦除，但是可能还是会导致部分数据损坏

raquelhortab, 2022-01-26:
There's no other way? I do have things I don't want to be lost...