JingOS-team/JingOS #100: Looking for japp-sdk-qt-dev package

onny, 2021-12-23:
**Describe the bug**
Hey, I would like to package JingOS apps for other Linux platforms. When trying to build an JingOS app following header file seems to be missing: `japplicationqt.h`.
Looking at the Debian package scripts, I guess the sdk dependency `japp-sdk-qt-dev` is most likely missing.
Is the source code for this already available?

Best regards
Jonas

JingOS-team, 2021-12-24:
Hey Jonas! I've assigned your request to the dev team. Our team member will get you back on workdays!
Have a merry Christmas!

songlijuan666, 2021-12-25:
这个需要王锐帮忙查看

JingOS-team, 2021-12-28:
Dear Jonas, 
Our team member has found out that japp package is obsolete and took it down for maintenance. The JingOS ARM 1.1.0 doesn't need those dependencies. If you need further help, please send us your question to service@jingos.com.

Kind Regards,
JingOS Team

closed 2021-12-28 by PaultheFish

PaultheFish, 2021-12-28:
Dear Jonas,

If you want to build the app on JingPad A1, please install the following item

```sudo apt install japp-sdk-dev japp-sdk-qt-dev```

However, if you want to build the app on other platform, there is no better way now.

onny, 2021-12-29:
Thank you for the fast response!

> However, if you want to build the app on other platform, there is no better way now.

Oh too bad. I see now that the library `libjappcoreqt.so` is present in the package `japp-sdk-qt` but only for `aarch64` :( Hopefully you can release the sources for the development packages soon :)