JingOS-team/JingOS #104: Add Widevine AArch64 support for Chromium/Firefox Web Browser on JingPad A1

Maxximo88, 2021-12-30:
Hi,
please find a way to get the library libwidevinecdm0 for AArch64 and run it on browsers of JingPad A1.
With that library users could open Netflix, Amazon Prime Video, Disney+ & co..

Thanks

FabioLolix, 2021-12-31:
Widevine does not exist yet for AArch64 and only Google can make an AArch64 build, it also have an unredestributable license

Maxximo88, 2022-01-03:
> Widevine does not exist yet for AArch64 and only Google can make an AArch64 build, it also have an unredestributable license

I fund it on Chrome OS build for AArch64..I saw the following topic dedicated to LuneOS & LuneBrowser but it does not work on JingPad A1:
https://pivotce.com/2017/10/11/luneos-browser-how-to-sideload-flash-widevine-plugins-netflix-amazon-youtube-red/
..But it explain how to..

damarte, 2022-01-03:
I have spent many hours on this problem and have not found a solution. 

I have tried to extract widevine from ChromeOS, from android, from raspberry pi and debian repositories, with chromium (snap and repository versions), vivaldi and firefox, nothing works.

As @FabioLolix says right now there is no version for ARM64 (there is for ARM32) and we don't have sources to compile, we depend on Google and that's usually not good news.

JingOS-team, 2022-01-06:
I'm sorry to inform you that we are unable to add this function due to the fact that this is a property of Google. We will keep looking for other solutions. Thanks for your support.

closed 2022-01-06 by JingOS-team