JingOS-team/JingOS #113: Japm - Multiple Issues after upgrade to 1.1.2

linuxrebel, 2022-01-21:
Issue 1 existing apps installed in 1.1.1 show up on desktop,  New apps being installed by Aurora store or similar no longer do. Apps appear to work, but only if you can find a back door way of launching 

Issue 2 If you launch a second android app.  It won't show up however if you swipe up, click the "android-app" icon it will switch to the app you just started.  Under 1.1.1 it would automatically display (will post video to this ticket if I can later)

Issue 3  if you close/swipe off the android-app icon or put it in the trash.  Then the only way to launch an android app is to first reboot the system.  

issue 4 if you have an Android app that starts automatically in the background (for example Twitter)  you cannot view that app nor can you launch a new one until you remove the app that is auto-starting and reboot.

linuxrebel, 2022-01-23:
Upgrade to 2.3.4 of japm 1.2.0 of jingos improved issue 2 a lot. Now each app has its own 'window'

Issue 3 also disappeared.

songlijuan666, 2022-01-24:
问题1：已知问题   应用商店安装后桌面不显示问题
问题2：已知问题   安卓应用多窗口问题
问题4：已知问题   安卓应用多窗口相关问题，应用无法查看后台启动多少应用
问题3：再尝试复现

samytichadou, 2022-01-25:
Adding my 2 cents here, tell me if this worth a new issue !

japm v2.3.4 (command `japm -v` returns `0.1.3`
android-compatible-env v1.0.5

- issue 5 : it is impossible to properly close an android app through ui. When closing it with the window manager (f5 button, then swipe app away), it does not appear anymore in the window manager, but htop still shows it, and a alt+tab shortcut makes it appear again. It is not closed, just hidden from the jingos window manager gui.