JingOS-team/JingOS #120: UI-Elements not displayed correctly on smaller screen

Gandalf1783, 2022-01-23:
**Describe the bug**
UI Elements just disappear without any slider or method to see its contents. 
Happens e.g. in the Installer when the Display-Resolution is too low.

**To Reproduce**
Steps to reproduce the behaviour:
1. Try to install JingOS via the 0.9 beta Installer
2. Display has to be relatively small (around 4:3 aspect ratio and around 1024px)

**Expected behaviour**
Some way to either scale the contents, so it's still readable but not too small, or some way of scrolling to the left/bottom to get all elements into the view.

**Screenshots**
[image]

**Additional context**
Cannot run the installer, since the buttons are hidden.
Can still reach them via TAB, but using this in a VM is impossible without upscaling the monitor settings.
(Will most likely happen IRL too when using devices with a smaller screen)

Gandalf1783, 2022-01-23:
Info: The resolution from the screenshot is around 640x480

PaultheFish, 2022-01-24:
We have stopped developing x86 version of JingOS. Thank you for pointing out

closed 2022-01-24 by PaultheFish