JingOS-team/JingOS #122: Upgrade flatpak package to somrhing >= 1.8.2

jjardon, 2022-01-23:
**Describe the bug**
flatpak version in the system (1.6.5) is too old to install some flatpaks (others work fine)
Current Debian and Ubuntu has already the up-to-date versions

**To Reproduce**
- $ flatpak install org.chromium.Chromium
error: Failed to install org.chromium.Chromium: app/org.chromium.Chromium/aarch64/stable needs a later flatpak version (1.8.2)

**Expected behavior**
Chromiuin flatpak should be possible to install

**Desktop (please complete the following information):**
 - OS: JingOS 1.2.0

jjardon, 2022-01-23:
ok, I tried to workaround the issue installing the flatpak from the ubuntu PPA: https://launchpad.net/~flatpak/+archive/ubuntu/stable

But I got another problem when trying to add it:
```
sudo add-apt-repository ppa:flatpak/stable      
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 108, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 118, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 613, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)    
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 91, in get_sources
    raise NoDistroTemplateException(
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Jingos/focal
```
After temporarily changing the `ID` in `/etc/os-release` from jingos to ubuntu everything works as expected:

Now I'm enjoying running chromium from a flatpak (which has also the advantage of being a more recent version (97.0.4692.99) than the one that comes with the device (93.0.4577.63))

Screenshot here:

songlijuan666, 2022-01-24:
问题已经反馈开发，我们内部分析一下