JingOS-team/JingOS #126: Unable to enter control characters in Terminal

jmaynard, 2022-02-26:
When using the Terminal application (Konsole), I am unable to enter any control characters at all from the virtual keyboard. I don't know if this is a keyboard or a Konsole issue.

To reproduce:
OPen Konsole by tapping on the dock icon.
In bash, try to enter control-C, either by tapping ctrl and then C, or else by holding ctrl and tapping C.

Expected: ^C appears on the command line and a new prompt is issued.
Actual: a single C is entered on the command line;

Environment: JingOS 1.2 on JingPad-C1 with JingOS loaded from the forum instructions.

jmaynard, 2022-02-26:
I just did a factory reset. No luck. Still doesn't let me use the control key.

writersglen, 2022-02-26:
> I have now tested with an external Bluetooth keyboard, and control characters work fine. THis is a virtual keyboard problem; I have revised the title accordingly.

Hi,

What do we know about the virtual keyboard software and developers? This is pretty serious issue. Perhaps we could incentive the developers to fix the bug by requesting directly. I’d even contribute to a bounty.

Best,

LRP