JingOS-team/JingOS #127: Install no working

Bowser04, 2022-03-09:
**Describe the bug**
during a launch and an installation the installer remains blocked/stuck during the copying of files And the mouse doesn't work anymore

**To Reproduce**
Get onda v919 air
Go go to boot menu with F7 to boot on jing os and try to install

**Expected behavior**
Nothing works anymore, it's blocked Since 1h30 of installation

**Screenshots**
I should put them on soon

**Desktop (please complete the following information):**
Material: onda v919 its x86 tablet

jmaynard, 2022-03-13:
That's not going to work. JIngOS is built for ARM processors, not x86.

Bowser04, 2022-03-13:
> That's not going to work. JIngOS is built for ARM processors, not x86.

No [image]

FabioLolix, 2022-03-13:
@Bowser04 last x86 image have been published in last year Spring, on 2021.12.14 they have announced that the x86 image is no longer developed and closed all related github tickets

Bowser04, 2022-03-13:
> @Bowser04 last x86 image have been published in last year Spring, on 2021.12.14 they have announced that the x86 image is no longer developed and closed all related github tickets

Ok thanks