JingOS-team/JingOS #129: Is the project Dead ?

MohamedAliRashad, 2022-04-30:
A final word from the maintainers of the project will be appreciated from everyone

onny, 2022-05-09:
unfortunately it never got completely open source so I couldn't reuse or package the apps for other distros :(

kuugan, 2022-05-12:
> unfortunately it never got completely open source so I couldn't reuse or package the apps for other distros :(

JingOS apps on other distros?

Probably not, but there should be enough to try working on them, just 0 documentation :p

ezrameow, 2022-07-13:
https://space.bilibili.com/2038810482
https://www.bilibili.com/video/BV1jq4y1y7Ci
The last video they uploaded on 01-12-2022 didn't mention any final words...
我猜他们倒闭了或者是跑路了...

kuugan, 2022-08-07:
> https://space.bilibili.com/2038810482 https://www.bilibili.com/video/BV1jq4y1y7Ci The last video they uploaded on 01-12-2022 didn't mention any final words... 我猜他们倒闭了或者是跑路了...
> 
> Their account's avatar is restored to default one so maybe this bilibili account is cancelled.

Yeah, that video was the 1.2 Update, shortly after everyone went missing :p

FabioLolix, 2022-09-01:
> Probably not, but there should be enough to try working on them, just 0 documentation :p

There is no source code to work on

ccchan234, 2023-02-16:
hi,

please rebuild your trust in chinese economic tablets by buying a chuwi hi10x or xr x86 tablet! just usd 200. (must use the intel/realtek sound chip variant)

then install fydeos (close source build on chromiumos, with android 9 support).

i dual boot win10+bitlocker with this chromeos alike OS, rocks daily.

it did support active stylus, and support pressure sensitivity/palm rejection in fydeos!

ps: don't buy chuwi hi10go, no touch screen.

btw, only the intel/realtek sound chip variant of hi10x/r is supported.
people say some variants use ESSxxxx, dont use that variant.