JingOS-team/JingOS #131: Download-server down?

jeremiaskleer, 2022-09-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

FabioLolix, 2022-09-01:
The project is dead

baifachuan, 2022-09-11:
> The project is dead

I want to get a Linux pad, if this pad is dead, any other pad can we choose?

rezazarchi, 2023-01-04:
Why is it dead? It has the best UI for touch devices and hybrid laptops.